med-material/ArduinoLogger #17: [R Shiny] Visualize Data from Physiological Tests

bastianilso, 2019-12-05:
This part is less important than issue #16  and #15 , but if we can make it, we would like to us the [ArduinoEDA-IBI-Pressure firmware](https://github.com/med-material/ArduinoEDA-IBI-Pressure) to capture people's Electrodermal Activity (measuring sweat conductivity) and heart rate (IBI) over time.  We want the captured data from the ArduinoLogger sent to an EDAIBIPressure table in the database and visualized by an R Shiny application as seen below:
[image]

Extra Notes:
- BPM and IBI needs to be calculated (how this can be done from knowing when the heart rate sensor has found a heart beat, more on that online).

hendrikknoche, 2019-12-05:
here's a reference screen shot from a tool called Kubios.
[image]

bastianilso, 2019-12-05:
Design todo:
- Have artifact threshold control for IBI/BPM
- Add meta-data (the parts which are useful) from Kubios
- Have a look at the EDA toolbox interface from Matlab

hendrikknoche, 2019-12-05:
here's something from an EDA analysis Matlab tool
[image]

it produces a list of events from this that can be grouped by predefined time windows (not shown).
[image]