med-material/ArduinoLogger #25: Create Standalone release for Windows, OSX, Linux

bastianilso, 2020-01-29:
In preparation for the upcoming RTII course, we want to distribute the ArduinoLogger as a standalone app, which the students can run on their own computers, [as have been done in the past](https://github.com/med-material/ArduinoLogger/releases).

For this release, we will need a few things:
- Update the READMEs of the [ReactionTime/SynchTest Firmware](https://github.com/med-material/ArduinoReactionSynchTests) and [EDA-IBI-Serial firmware](https://github.com/med-material/ArduinoEDA-IBI-Pressure) with step-by-step instructions on how to use them with the ArduinoLogger (including screenshots and highlighting what button to press and sensors is in use). Include a link which the students can use to check the R Shiny Data. Emphasize what to do if something goes wrong (e.g. make sure you are on the AAU Wi-Fi). [See a reference step-by-step guide used last year here.](https://docs.google.com/document/d/1jxZlyuChqvD5O-oMInEq9d3nAXqxzNDJT5h9wjeJnPA/edit#heading=h.i9g7g9mvftbu) and [here](https://docs.google.com/document/d/1Cwi2ZenfqutQLCzDGr4BaUXwRIZBr09N7QQHQ__48P4/edit#heading=h.i9g7g9mvftbu).
- For the SynchTest/ReactionTime instructions, we need to explain how to change between ERM/LED and how to set MusicalAbility (to values HIGH or LOW).
- Assign the ArduinoLogger a new version number and write a short list of what changed. Export from Unity to OSX, Linux, Windows.
- Test the exported versions on the respective platforms, with the different arduino firmwares and check that the data gets sent to the database and can be found in R  Shiny.

bastianilso, 2020-01-29:
Additionally when releasing, create a commit which updates the ArduinoLogger README indicating which version is the latest and linking to the release page.

MaximeHollebecq, 2020-01-30:
the Mac release on the previous release doesn't work.

bastianilso, 2020-01-30:
@MaximeHollebecq it should work - it was tested before it was released. If it doesnt work now, it may be because OS X updated or similar. What parts about it exactly doesnt work? Are you able to make a working Mac release from master?