med-material/ArduinoLogger #32: mac error "unknown column TimeStamp" - new release needed?

hendrikknoche, 2020-03-19:
On a MAC when trying to publish logs to the server I get the below error.
I think we might have changed the table structure and the MAC release didn't get updated.
[image]

bastianilso, 2020-03-19:
The issue was in the database - we had no column for collecting timestamps for EDAIBI-Serial data. I added the column now. Could you try again?

hendrikknoche, 2020-03-20:
one more field missing :-)
[image]

bastianilso, 2020-03-20:
Did you add it yourself? I can see there is a column called Millis inside the database.

hendrikknoche, 2020-03-20:
> Reply to: med-material/ArduinoLogger <reply@reply.github.com>

Hmm, strange. I think I might have added it at some point but it still throws the error

best

Hendrik

hendrikknoche, 2020-03-20:
it seems the error is on the database side, right?

bastianilso, 2020-03-20:
@hendrikknoche could you save it to CSV and upload it to this thread?
very peculiar, cause we do have the column there,..
`[Fri Mar 20 08:41:52.229286 2020] [:error] [pid 39339] [client 172.29.53.237:65462] prepare() failed: Unknown column 'Millis' in 'field list'`

hendrikknoche, 2020-03-20:
here it is... it's the email most likely

Email;Millis;EDA;IBI;RawPulse;Pressure;Button
;
2020-03-20 11:48:18;hk@create.aau.dk;1;1;1857;0;0;81;0;0;
2020-03-20 11:48:18;hk@create.aau.dk;1;1;1867;4;0;80;0;0;
2020-03-20 11:48:18;hk@create.aau.dk;1;1;1878;0;0;81;0;0;
2020-03-20 11:48:18;hk@create.aau.dk;1;1;1888;4;0;81;0;0;
2020-03-20 11:48:18;hk@create.aau.dk;1;1;1898;0;1894;80;0;0;

bastianilso, 2020-03-20:
Could you upload the CSV file itself? as an attachment (drag n drop)? sorry, i need to be sure I understand what is in what column

hendrikknoche, 2020-03-20:
[RTIIlog.csv.zip](https://github.com/med-material/ArduinoLogger/files/4359258/RTIIlog.csv.zip)

bastianilso, 2020-03-20:
[image]

It looks to me like there is something wrong with the output here + some minor unrelated issues:
- When exporting to CSV files, the ArduinoLogger does not insert a "TimeStamp" value in the header for column 1 (however, this is not a problem for DB logging, thats for sure).
- All columns except "E-mail" column, seem to be mismatched to their header.
- There is a surplus of columns, compared to headers.

The problem we see with columns mismatching/being too many, is likely the source of the DB logging problem. I have updated our php file with some extra debug logging.
Could I ask you to do the following:

- Remove any "coldump" and "logdump" files present inside `~/Library/Application Support/AAU/ArduinoLogger/Data` (alternatively, try Ctrl+F for coldump/logdump on your Mac and remove any that appears).
- Open Arduino IDE and inside it, the Serial Monitor, then check that there are only 6 columns present in the output?
- Open the ArduinoLogger software, record some data, try to publish it, save as CSV and upload the CSV here?

Then i will make a comparison and see what is happening. If all of this goes wrong, we might want to try some remote-control over MS teams and debugging in Unity.

hendrikknoche, 2020-03-20:
let's do a screen share call around this. I cannot find that directory (after AAU it goes to RTII and there's no Data subdir). I'm getting another error 
[image] PID this time.
Was I supposed to DL a new ino or dmg file?
[log.csv.zip](https://github.com/med-material/ArduinoLogger/files/4360819/log.csv.zip)

bastianilso, 2020-03-23:
It seems like the main issue was the missing `PID` column in the database. The database uploading works now.

closed 2020-03-23 by bastianilso