med-material/ArduinoLogger #34: on mac the logger throws an error

hendrikknoche, 2020-03-25:
[image]

hendrikknoche, 2020-03-25:
but does actually write data successfully after to the db

bastianilso, 2020-03-26:
interesting.. "insert into synch" ?
Could you test whether you get the same error message if you upload the ReactionTime exercise to the Arduino and publish to server?

bastianilso, 2020-03-26:
I can explain that I believe it happens because the default target is `synch` - which does not have a `Millis` column. But why exactly our code is sending data to `synch` is still a mystery to me.

Given that you are getting this error at the very beginning (prior to recording), I believe the data being sent is dumped invalid data from an earlier sessions that ConnectToMySQL is trying to re-upload. Could you check if "logdump" or "coldump" files exist in one of these filepaths?

`/Library/Application Support/company name/product name`

`~/Library/Caches`

`~/Library/Application Support/unity.company name.product name`

hendrikknoche, 2020-03-26:
I found three files:

[Archive.zip](https://github.com/med-material/ArduinoLogger/files/4387431/Archive.zip)
deleted them from my machine now.

bastianilso, 2020-03-26:
@hendrikknoche ah, I realize what the problem is now. When we dump logs due to failure, we do not dump information about which table the logs are supposed to go into (because we only support sending to a single table, arduino logger is just changing the table ad-hoc). So they just get sent to the default table which happens to be `synch`.