med-material/ArduinoLogger #50: Arduino cannot connect to port (Version 2022.02.17)

mathiassand, 2022-03-16:
MacOS High Sierra, Version 10.13.6

Throws error when trying to connect: "Could not connect to Arduino on port: /dev/cu.usbmodem1421".

On the following version I was able to connect to the arduino, but it gave a DB Error "Could not reach the database": Version 2020.04.06_MAC (under v2020.02.10 Universal Arduino Logger)

hendrikknoche, 2022-03-16:
same problem on Mac OSX 10.15 but connected through USB hub

mathiassand, 2022-03-22:
https://forum.unity.com/threads/libmonoposixhelper-dylib-in-2019-3-with-workaround.872632/

Getting error: libMonoPosixHelper.dylib

To fix this error, ctrl + right click the downloaded ArduinoLogger_OSX_version and:
1) press "Show Package Contents".
2) Go into the frameworks folder
3) Create (inside the frameworks folder) the following directories:

MonoBleedingEdge/MonoEmbedRuntime/osx

4) move libMonoPosixHelper.dylib from the frameworks folder to the osx folder.